uTensor/uTensor #1: build failed

Knight-X, 2017-09-24:
command: mbed compile -t ARM -m K64F
what is target board ? 
ARM_PATH = "/usr/local/ARM_Compiler_5.06u5"

[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /root/2/mbed-os/tools/make.py -t ARM -m K64F --source . --build ./BUILD/K64F/ARM" in "/root/2"

neil-tan, 2017-09-24:
A Readme.md is added with build instruction. Could you please post a more comprehensive error message if it still fails?

neil-tan, 2017-09-24:
If the build still fails, could you use GCC?
https://launchpad.net/gcc-arm-embedded

This project uses C++11, in case ARM Compiler does not support it.

Knight-X, 2017-09-25:
Building project uTensor (K64F, ARM)
Scan: .
Scan: env
Scan: mbed
Scan: FEATURE_STORAGE
Scan: FEATURE_LWIP

[OS ERROR] Command: /usr/local/ARM_Compiler_5.06u5/bin/armcc -c --gnu -Ospace --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 -DNDEBUG --cpu=Cortex-M4.fp --md --no_depend_system_headers --c99 -D__ASSERT_MSG --cpp --no_rtti --no_vla -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DTARGET_KPSDK_MCUS -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DDEVICE_SPI_ASYNCH=1 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DDEVICE_ERROR_RED=1 -DMBED_BUILD_TIMESTAMP=1506320876.72 -DDEVICE_TRNG=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTARGET_FRDM -DTARGET_MCUXpresso_MCUS -DDEVICE_SPI=1 -DFEATURE_STORAGE=1 -DTOOLCHAIN_ARM_STD -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DDEVICE_PORTIN=1 -DTOOLCHAIN_ARM -DTARGET_MCU_K64F -DARM_MATH_CM4 --via ./BUILD/K64F/ARM/.includes_ffee9d40a223382ea8135e13ebf5146b.txt --preinclude=./BUILD/K64F/ARM/mbed_config.h --depend ./BUILD/K64F/ARM/./main.d -o ./BUILD/K64F/ARM/./main.o ./main.cpp
[OS ERROR] Command: /usr/local/ARM_Compiler_5.06u5/bin/armcc -c --gnu -Ospace --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 -DNDEBUG --cpu=Cortex-M4.fp --md --no_depend_system_headers --c99 -D__ASSERT_MSG --cpp --no_rtti --no_vla -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DTARGET_KPSDK_MCUS -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DDEVICE_SPI_ASYNCH=1 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DDEVICE_ERROR_RED=1 -DMBED_BUILD_TIMESTAMP=1506320876.72 -DDEVICE_TRNG=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTARGET_FRDM -DTARGET_MCUXpresso_MCUS -DDEVICE_SPI=1 -DFEATURE_STORAGE=1 -DTOOLCHAIN_ARM_STD -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DDEVICE_PORTIN=1 -DTOOLCHAIN_ARM -DTARGET_MCU_K64F -DARM_MATH_CM4 --via ./BUILD/K64F/ARM/.includes_ffee9d40a223382ea8135e13ebf5146b.txt --preinclude=./BUILD/K64F/ARM/mbed_config.h --depend ./BUILD/K64F/ARM/mbed-os/drivers/BusIn.d -o ./BUILD/K64F/ARM/mbed-os/drivers/BusIn.o ./mbed-os/drivers/BusIn.cpp
...(The same message over and over)
[OS ERROR] Command: /usr/local/ARM_Compiler_5.06u5/bin/armcc -c --gnu -Ospace --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 -DNDEBUG --cpu=Cortex-M4.fp --md --no_depend_system_headers --c99 -D__ASSERT_MSG -DFEATURE_LWIP=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DTARGET_Freescale -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DFSL_RTOS_MBED -DDEVICE_STORAGE=1 -DTARGET_KPSDK_MCUS -DTARGET_CORTEX_M -DTARGET_KSDK2_MCUS -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_K64F -DDEVICE_SPI_ASYNCH=1 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DCPU_MK64FN1M0VMD12 -DTARGET_LIKE_MBED -DTARGET_FF_ARDUINO -DTARGET_KPSDK_CODE -DTARGET_RELEASE -DDEVICE_SERIAL_FC=1 -DDEVICE_ERROR_RED=1 -DMBED_BUILD_TIMESTAMP=1506320876.72 -DDEVICE_TRNG=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTARGET_FRDM -DTARGET_MCUXpresso_MCUS -DDEVICE_SPI=1 -DFEATURE_STORAGE=1 -DTOOLCHAIN_ARM_STD -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DDEVICE_PORTIN=1 -DTOOLCHAIN_ARM -DTARGET_MCU_K64F -DARM_MATH_CM4 --via ./BUILD/K64F/ARM/.includes_ffee9d40a223382ea8135e13ebf5146b.txt --preinclude=./BUILD/K64F/ARM/mbed_config.h --depend ./BUILD/K64F/ARM/mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/netif/ppp/lwip_mppe.d -o ./BUILD/K64F/ARM/mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/netif/ppp/lwip_mppe.o ./mbed-os/features/FEATURE_LWIP/lwip-interface/lwip/src/netif/ppp/lwip_mppe.c

mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /root/uTensor/mbed-os/tools/make.py -t ARM -m K64F --profile ./build_profile/release.json --source . --build ./BUILD/K64F/ARM" in "/root/uTensor"

neil-tan, 2017-09-25:
This seem to be more of a tool-chain issue than the repository's.
Could you please verify that `$ARM_PATH\bin\armcc` is reachable?

This might be relevant:
https://os.mbed.com/questions/73880/Get-error-whil-compiling-mbed-os-example/

Knight-X, 2017-09-25:
root@0e29ace78b3c:~/uTensor# mbed compile -t ARM -m K64F --profile=./build_profile/release.json
usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG] [-p PROGRAM]
               [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS]
               [-S [{matrix,toolchains,targets}]] [-f GENERAL_FILTER_REGEX]
               [--stats-depth STATS_DEPTH] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rpc] [--usb] [--dsp]
               [--testlib] [--build-data BUILD_DATA] [-l LINKER_SCRIPT]
make.py: error: Could not find executable for ARM.
Currently set search path: /usr/local/ARM_Compiler_5.06u5/bin64
[mbed] ERROR: "/usr/bin/python" returned error code 2.
[mbed] ERROR: Command "/usr/bin/python -u /root/uTensor/mbed-os/tools/make.py -t ARM -m K64F --profile ./build_profile/release.json --source . --build ./BUILD/K64F/ARM" in "/root/uTensor"

neil-tan, 2017-09-25:
@Knight-X has communicated that lib32stdc++6 needs to be installed. No further testing were carried out as a ARM-compiler license is required.

closed 2017-09-25 by neil-tan