uTensor/uTensor #3: compiletestdata.py error in linux

Knight-X, 2017-09-28:
1. 
  Traceback (most recent call last):
  File "compileTestData.py", line 28, in <module>
    FULL_OUTPUT_DIR = (Path.cwd() / OUTPUT_DIR).resolve()
  File "/usr/lib/python3.5/pathlib.py", line 1109, in resolve
    s = self._flavour.resolve(self)
  File "/usr/lib/python3.5/pathlib.py", line 330, in resolve
    return _resolve(base, str(path)) or sep
  File "/usr/lib/python3.5/pathlib.py", line 315, in _resolve
    target = accessor.readlink(newpath)
  File "/usr/lib/python3.5/pathlib.py", line 422, in readlink
    return os.readlink(path)
FileNotFoundError: [Errno 2] No such file or directory: '/root/uTensor/TESTS/scripts/testData'

2. Traceback (most recent call last):
  File "compileTestData.py", line 47, in <module>
    exec(open(str(item), "r").read())
  File "<string>", line 17, in <module>
  File "compileTestData.py", line 26, in mkdir
    os.makedirs(outPath)
  File "/usr/lib/python3.5/os.py", line 226, in makedirs
    head, tail = path.split(name)
  File "/usr/lib/python3.5/posixpath.py", line 103, in split
    i = p.rfind(sep) + 1
AttributeError: 'PosixPath' object has no attribute 'rfind'

Knight-X, 2017-09-28:
1.  before call path.resolve(), the folder should be existed in linux environment. Otherwise, it would output error message.

2. the parameter in os module should be str

neil-tan, 2017-09-28:
Thanks for the PR, merged.

closed 2017-09-28 by neil-tan