uTensor/uTensor #31: documentation on training and tensorflow export

sandys, 2017-11-05:
can you update your documentation to elaborate more on how to do training and export and use the data in the microcontroller.

Only thing current github project mentions is "The related Tensorflow training script please refer to the node-viewer project". Its very disconnected and we are not able to connect them both properly.

neil-tan, 2017-11-05:
Hey.  Definitely.

  Although, for the next 2 weeks, the focus is on getting the release 0.1 out of oven, I can give you some foresight here.
### The basic flow is:

  - Train your model (Pay attention to what Ops you use. There is only basic ones like MatMul, Relu, Argmax, Quantization,etc )
  - Freeze you graph
  - Quantize the graph
  - Using Tensor Board, Identify all the constants (Tensors) you'd have to export
  - Use tf-node-viewer to export the constants (Tensors) to idx files
  - To the device

  There are changes coming up soon to make this easy, and there will be blogs about this. Please only treat the current project as a preview.

mbartling, 2017-11-20:
Hey @sandys, I am working on the tooling for squashing this flow into:

- Train your model
- Freeze the graph (ex. to a .pb file)

Running utensor-cli will handle the optional quantization, IDX dumps, and interface creation on the device.

You can find the current updates here: https://github.com/mbartling/utensor-cli

neil-tan, 2018-02-03:
Release 0.2 is live and should address this.

closed 2018-02-03 by neil-tan