uTensor/uTensor #57: OverflowError: Python int too large to convert to C long

rkuo2000, 2017-11-21:
// python3.6.3 64bit installed on Windows 7 Ultimate SP1 64bit 
$ python compileTestData.py
FULL_OUTPUT_DIR C:\Users\RKUO\uTensor\TESTS\scripts\testData
FULL_PREGEN_DIR C:\Users\RKUO\uTensor\TESTS\scripts\PRE-GEN
float_4d_power2 sum:  -1.0
int16_4d_power2 sum:  270250
Traceback (most recent call last):
  File "compileTestData.py", line 48, in <module>
    exec(open(str(item), "r").read())
  File "<string>", line 12, in <module>
OverflowError: Python int too large to convert to C long

Knight-X, 2017-11-21:
thanks for your report. @neil-tan The bound of tmp variable in int32_4d_power2.py sould be less than 2 ** 31 not 2 ** 32. I will fix this.

neil-tan, 2017-11-26:
Thanks to @Knight-X, this issue has been addressed. Please pull, update testData and run again.

closed 2017-11-26 by neil-tan