uTensor/uTensor #93: Public Channel

neil-tan, 2018-04-13:
We need a way to interact with external developers as the project continue to grow. The public channel let us:

- Collect Feedback
- Support
- Bring transparency to our dev cycles
- Archive Q&A and design decision

We currently have a google emailing list, but is there something more we can do?

We have a public google group with slack integration. Can we do something that has a shorter turn-around time than emails and is google-searchable? Current ideas:

- Make Slack Channel Public (@mbartling)
- Make the Slack channel public and mirror to [Gitter](https://gitter.im) via a [Sameroom](https://sameroom.io/blog/connecting-a-channel-in-slack-to-a-room-in-gitter/)
- Make a forum
- Or, something else

[Project link](https://github.com/orgs/uTensor/projects/1)

neil-tan, 2018-04-13:
Using only google-group, what we have now, is also an option.

nimaaghli, 2018-04-13:
Can you please provide your google group link?

neil-tan, 2018-04-14:
Hey @nimaaghli 
It is here:
https://groups.google.com/forum/#!forum/utensor
utensor@googlegroups.com

neil-tan, 2018-05-02:
Guys, ever get that feeling of wanting a new toy? I just tried out gitter and thought it's kinda nice. You can readily use it if you have a GitHub account which is quite relevant to our audiences.

It's too much hassle to move away from Slack at this point. Though, gitter may work well as a public channel. All we have to do is to integrate it with our Slack.

Then, there's the question of google-group vs slack, or both. What do you guys think?