uTensor/uTensor #96: binary/inline tensor

Knight-X, 2018-04-21:
To avoid exploration of parameter of layers, we could compile weight parameter into flash device. We could probably make those parameters read only flash area and create tensors by const array instead of  idxImporter. Any Ideas ?

mbartling, 2018-04-23:
I think we should still maintain the importer layer.
 i.e. something like this, or have an importer factory or something.

```
class ImporterBase{
 public:
   virtual some_read_write_interface(...)
};
class ImporterIdxSD : public ImporterBase ...
class ImporterIdxConst: public ImporterBase ...
```

Knight-X, 2018-04-24:
That is cool and it will minimize the complexity of idxloader now.  Just we have to consider whether we support idx format or not ? In order to minimize the code size, i would prefer to store the const weight pointer purely. How do you think ?

neil-tan, 2018-04-25:
Performance is a major consideration for the binary/inline tensor class. It would be nice if it adapts target devices' native format.

I am trying to think of the reason for keeping the importer here. @mbartling, my guess is we can do more runtime trick this way?

Keep in mind that every tensor also inherits an init and de-init hook from the tensor-base class.

mbartling, 2018-04-26:
I would still sacrifice the slight read time and have the `FormatReaders` separate from the `PlaceImporters`. 
For example `ImporterSD<IdxReader> myReader();` or `ImporterSD<WAVReader>`, or `ImporterSdRam<ConstReader>`. 

Block memory reads are way more expensive than adding one more read pointer function call, so having the additional configurability greatly simplifies the code base without any real cost. Additionally, it makes adding Exporters trivial. 

@Knight-X what do you think

neil-tan, 2018-04-26:
I like the idea of having the `FormatReaders` and `PlaceImporters` separated. We might be able to make universal tensor importers using polymorphism, `Importer(Tensor* t, FmtReader* pipe)`

@mbartling Can you clarify what you meant by "Block memory"? Block in the sense of a chunk of memory or busy I/O?

neil-tan, 2018-04-27:
wave files, think of it as bmp for sound.

mbartling, 2018-04-27:
@neil-tan Block memory as grab adjacent bytes. It's partially file system dependent for performance

neil-tan, 2018-04-29:
To give it a bit of context:
- The Tensor class [initialization](https://github.com/uTensor/uTensor/blob/8fde39d05bc20c82fa7ca4499b7ad531e7ec6fe0/uTensor/core/tensor.hpp#L117)
- [Heap allocation](https://github.com/uTensor/uTensor/blob/8fde39d05bc20c82fa7ca4499b7ad531e7ec6fe0/uTensor/core/tensor.hpp#L126)
- Different types of tensors can be derived from the base Tensor, for example, the [SDTensor](https://github.com/uTensor/uTensor/blob/8fde39d05bc20c82fa7ca4499b7ad531e7ec6fe0/uTensor/core/sdtensor.hpp#L14).

- The importers are [here](https://github.com/uTensor/uTensor/tree/master/uTensor/loaders).

Knight-X, 2018-04-29:
I think that clean importer interface is nice, but we need to consider whether it could reduce the sram memory usage in binary tensor. Should we just consider the interface of making binary tensor first ? Then, we could just make importer interface clean and decide what type of tensor should be supported.

neil-tan, 2018-04-29:
Hey, just spoke with Marcus, looks like the bottomline is having the tensor-data-void-pointer be static-const. This should allow us to keep the model parameters on the flash instead of heap. Creating a new Tensor class is probably the easiest way. It will be read-only. Don't mind having the importer here as long as the interface makes sense and keeps the data in flash.

marcuschangarm, 2018-04-29:
> the bottomline is having the tensor-data-void-pointer be static-const.

Just to clarify, the important part is having the actual array being declared `const` and not just the pointer.

Knight-X, 2018-05-02:
I have drafted binary tensor, and i will write test case which is similar to sdtensor test for it. Therefore, I need some feature to transform weights to const pointer in c++, which will be in codegen.

chynphh, 2018-09-19:
uTensor is a amazing job.  I want to use it to speed up my model. But I got some trouble.
How can I initialize Tensor with a vector or other something, not idx file？Just like this for 
"simple_mnist":

`    
    Tensor *data; 
    data = new Tensor();
    std::vector<uint32_t> tmp(784, 0);
    data->init(tmp);
    get_simple_mnist_ctx(ctx, data);
    S_TENSOR pred_tensor = ctx.get("y_pred:0");
    ctx.eval();
 
    print_tensor<int>(pred_tensor);
`
Thank you a lot.

dboyliao, 2018-09-19:
If you want to initialize tensor with zeros, how about:
```
uint32_t *w_ptr = data->write<uint32_t>(0, 0);
for (uint32_t i = 0; i < data->getSize(); ++i) {
    *(w_ptr + i) = 0;
}
```
Or you want sth without a loop?

neil-tan, 2019-12-11:
done

closed 2019-12-11 by neil-tan