uTensor/uTensor #98: Release 0.25

neil-tan, 2018-04-21:
Version 0.2.0 being the previous release. The master branch is quite behind of the develop branch. Many useful features has been implemented since then. Perhaps it's time for a half-release, version 0.2.5. Let's use this issue to discuss what's to be include in this version.

neil-tan, 2018-04-21:
What do you think of the version number? We might want to include CNN support in version 0.3.

mbartling, 2018-04-23:
Dont forget to add unified testing structure.

neil-tan, 2018-04-26:
lol  guys
@dboyliao All there's missing is dropout, then, you'd have a full-blown CNN. Should we add it in?
@mbartling yeah  that's going into the release note

dboyliao, 2018-04-26:
I just finish the dropout name scope parsing thing tonight.
I think dropout is working but I need more tests to be sure.