uTensor/uTensor #103: Update Tensor R/W interface

mbartling, 2018-06-09:
Proposed update for the Tensor read write interface:

```cpp
// Currently
template <class T>
const T* read(size_t offset, size_t ele) {
    return (const T*)read(offset, ele);
}
...
const float input_min = in_min->read<T2>(0, 0)[0]; //Should be const float input_min = *(in_min->read<T2>(0, 0)); because this makes more sense

// Proposed
template <typename T>
const T& read(size_t offset, size_t ele) const {
    return (const T&)(*read(offset, ele));  
}

// Pretty sure this just needs a decltype somewhere to work
template <typename T>
auto read<T*>(size_t offset, size_t ele) const -> decltype(T*) {
    return (const T*)read(offset, ele); 
}

// Example use
const float input_min = in_min->read<T2>(0, 0);
const float* data = m_data->read<float>(0, 0);

// For linear indexing
template <typename T>
const T& Tensor<T>::operator[](size_t offset) const {
    return this->read(offset, sizeof(T));
}

template <typename T>
T& Tensor<T>::operator[](size_t offset) {
    return this->read(offset, sizeof(T));
}
```

neil-tan, 2018-06-10:
@Knight-X Did you mean we should divide the work among us to update the tensor interface for all operator classes?

Knight-X, 2018-06-11:
ya, thanks for your explanation. Really precise.