uTensor/uTensor #119: TypeError: ("'backend' must be <type 'unicode'> (got 'tensorflow' that is a <type 'str'>)."

tctchris, 2018-07-16:
Hi,

I get the following error when I run ` utensor-cli mnist_model/deep_mlp.pb --output-nodes=y_pred`:

WARNING:tensorflow:From /usr/local/lib/python2.7/site-packages/tensorflow/contrib/learn/python/learn/datasets/base.py:198: retry (from tensorflow.contrib.learn.python.learn.datasets.base) is deprecated and will be removed in a future version.
Instructions for updating:
Use the retry module or similar alternatives.
2018-07-16 11:41:24.088543: I tensorflow/tools/graph_transforms/transform_graph.cc:264] Applying sort_by_execution_order
Traceback (most recent call last):
  File "/usr/local/bin/utensor-cli", line 11, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python2.7/site-packages/utensor_cgen/__main__.py", line 79, in cli
    main(**args)
  File "/usr/local/lib/python2.7/site-packages/utensor_cgen/__main__.py", line 33, in main
    generator.generate(model_path)
  File "/usr/local/lib/python2.7/site-packages/utensor_cgen/code_generator.py", line 44, in generate
    self._generate_from_pb(src_fname)
  File "/usr/local/lib/python2.7/site-packages/utensor_cgen/code_generator.py", line 65, in _generate_from_pb
    ugraph = uTensorGraph(graph_def, self.output_nodes)
  File "/usr/local/lib/python2.7/site-packages/utensor_cgen/ir/base.py", line 148, in __init__
    self._init_from_graph_def(graph, output_nodes)
  File "/usr/local/lib/python2.7/site-packages/utensor_cgen/ir/base.py", line 253, in _init_from_graph_def
    op_attr=op_attr)
  File "<attrs generated init d9962e57b1b01374ac28f3b916c32d4d22a17054>", line 16, in __init__
  File "/usr/local/lib/python2.7/site-packages/attr/_make.py", line 1668, in __call__
    v(inst, attr, value)
  File "/usr/local/lib/python2.7/site-packages/attr/validators.py", line 33, in __call__
    attr, self.type, value,
TypeError: ("'backend' must be <type 'unicode'> (got 'tensorflow' that is a <type 'str'>).", Attribute(name='backend', default=NOTHING, validator=_AndValidator(_validators=(<instance_of validator for type <type 'unicode'>>, <function check at 0x7f37ed262668>)), repr=True, cmp=True, hash=None, init=True, metadata=mappingproxy({}), type=None, converter=None), <type 'unicode'>, 'tensorflow')

How can I fix it?
Has it something to do with Python2 Python3 unicode [More About Unicode in Python 2 and 3
](http://lucumr.pocoo.org/2014/1/5/unicode-in-2-and-3/)?

I am using the Cloud9 environment.

Thanks

dboyliao, 2018-07-17:
Yes.
It seems like I used the wrong text type.
Thanks for reporting this bug.

dboyliao, 2018-07-17:
@tctchris Can you file this PR in the repo of the utensor cli?
It's over here: https://github.com/uTensor/utensor_cgen

dboyliao, 2018-07-17:
@tctchris 
If you want to fix it by yourself, the bug is over here:
https://github.com/uTensor/utensor_cgen/blob/develop/utensor_cgen/ir/base.py#L94
and 
https://github.com/uTensor/utensor_cgen/blob/develop/utensor_cgen/ir/base.py#L252

Maybe you'd like to take a look on `attrs` first:
http://www.attrs.org/en/stable/examples.html

I can fix it only until today night.
Just in case if you can't wait for me.
Besides, it will take even more time for us to update the cloud 9 environment.

dboyliao, 2018-07-18:
@tctchris The bug is fixed and the master branch of `utensor_cgen` has been updated.
Closing this issue

closed 2018-07-18 by dboyliao

dboyliao, 2018-07-18:
@tctchris  BTW, I've upload new version to PyPI.
`utensor-cli` version 0.2.3 should work.
You can run `utensor-cli -v` to see the version on your machine.

tctchris, 2018-07-18:
It worked with the Cloud9 workflow, thanks!