uTensor/uTensor #129: Is it possible to use it with MicroPython

j16180339887, 2018-08-27:
See also 

https://github.com/micropython/micropython 

> MicroPython implements the entire Python 3.4 syntax (including exceptions, with, yield from, etc., and additionally async/await keywords from Python 3.5).

https://github.com/micropython/micropython/wiki/FAQ

> 128K ROM/8K RAM is the recommended minimum configuration for doing something "real".

Thanks!

Knight-X, 2018-08-28:
That is a very interesting idea. Although I am not familiar with micropython, I think that we have some ways to integrate with it.  Intuitively, you need to write the interface between micropython and its corresponding c++ ops. Take a look at our ops directory and give it a try if interested. https://github.com/uTensor/uTensor/tree/master/uTensor/ops

joeky888, 2018-08-28:
https://github.com/joeky888/awesome-micropython-lib

Actually I was searching for all MicroPython libs,

and I am new to MicroPython as well.

> to write the interface between micropython and its corresponding c++ ops

I am not sure if libffi can do that
 https://github.com/micropython/micropython/blob/master/examples/unix/ffi_example.py

Every MicroPython library I've found were written in pure python code. 

I'll give it a try.

torntrousers, 2018-10-31:
Did you ever get anywhere with this?

joeky888, 2018-10-31:
No, I was working on embedded linux and having troubles to cross-compile some Python libs.

After that I switch to Go, Go is extremely easy to cross-compile code to a single static binary on any platform (if all libraries are written in pure Go).

then I use [upx](https://github.com/upx/upx) to compress the binary (reduce binary size to 25%) and put it on my board.

Didn't even try uTensor, I am very sorry about this.