uTensor/uTensor #134: test case code refactoring

Knight-X, 2018-10-09:
In order to simplify the steps of writing test case, we probably need to do some refactoring. For example, we need to run tensorflow and generate idx file manually now. I think that we can write scripts to generate data before running test case automatically. 
Then, we can just compare the result between tensorflow and utensor. It will be easier to extend other test case to increase test coverage.

mbartling, 2018-10-09:
@Knight-X The only test cases that truly need IDX files are the tensor ops tests and IDX loader. These are assumed to be fixed mathematical operators with well defined behavior as a function of their inputs.

neil-tan, 2018-10-09:
@Knight-X @mbartling @dboyliao 
Guys, while we are at it, we should replace idx file with ROM_tensor.

Knight-X, 2018-10-12:
rom_tensor is fine. But we still need a tool to generate weight array or we have to do it manually. Compared to pushing weight files on github, I think that generating the necessary weight files before running all tests make more sense.