uTensor/uTensor #136: Error when compiling utensor-mnist-demo

drajmund, 2018-11-11:
I followed the instruction, and tried to compile the utensor-mnist-demo with the command:
mbed compile -m DISCO_F413ZH -t GCC_ARM --profile=uTensor/build_profile/develop.json

I received the following message with an error:

Building project utensor-mnist-demo (DISCO_F413ZH, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Compile [  4.9%]: main.cpp
[Error] image.h@7,50: expected 'class' before 'TENSOR'
[ERROR] In file included from ./main.cpp:6:0:
./image.h:7:50: error: expected 'class' before 'TENSOR'
 template<typename T, template<typename> typename TENSOR=RamTensor>
                                                  ^
[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/home/drajmund/utensor-mnist-demo"
       Command: "/usr/bin/python -u /home/drajmund/utensor-mnist-demo/mbed-os/tools/make.py -t GCC_ARM -m DISCO_F413ZH --profile uTensor/build_profile/release.jso
n --source . --build ./BUILD/DISCO_F413ZH/GCC_ARM-RELEASE"
       Tip: You could retry the last command with "-v" flag for verbose output
---

Do you have any idea what is wrong with it?
System parameters:
Ubuntu 16.04.5 LTS
Python 2.7.12
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
mbed-cli version number: 1.8.2

neil-tan, 2018-11-11:
Hey @drajmund 
You might want to update your gcc to version 5 or 6 as described in the [mbed-cli doc](https://github.com/ARMmbed/mbed-cli#compiling-code)

Once you got it working, you might also be interested in this [branch](https://github.com/uTensor/utensor-mnist-demo/tree/f/rom-tensor) which does not require an SD card. You can do this by using `git checkout f/rom-tensor`, `mbed deploy` follow by a clean compile.

Please let us know how it goes!

dboyliao, 2018-11-11:
BTW, we haven't fix the include order yet.
That is, `context.hpp` must be included first before other header files.
I know I can fix it in the `cli` but I don't think it's a good approach and leaved it as is then.

dboyliao, 2018-11-11:
Please refer to #65 for detail.

drajmund, 2018-11-11:
Thank you for your help.

I downloaded the new version of gcc-arm-none-eabi and configured mbed as follows:
$ mbed config -G GCC_ARM_PATH /usr/local/gcc-arm-none-eabi-6-2018-q2-update/bin

I added the support for BSP_DISCO_F413ZH:
$ mbed add https://os.mbed.com/teams/ST/code/BSP_DISCO_F413ZH/
$ mbed deploy

Compilation is OK, but I receive "multiple definition of" errors from the linker:

$ sudo mbed compile -m DISCO_F413ZH -t GCC_ARM --profile=uTensor/build_profile/release.json -c
Building project utensor-mnist-demo (DISCO_F413ZH, GCC_ARM)
Scan: utensor-mnist-demo
Scan: env
Compile [  0.1%]: ft6x06.c
Compile [  0.3%]: st7789h2.c
Compile [  0.4%]: wm8994.c
...
Compile [ 99.7%]: tensorIdxImporter.cpp
Compile [ 99.9%]: quantization_utils.cpp
Compile [100.0%]: uTensor_util.cpp
Link: utensor-mnist-demo
BUILD/DISCO_F413ZH/GCC_ARM-RELEASE/F413ZH_SD_BlockDevice/BSP_DISCO_F413ZH/Drivers/BSP/Components/ft6x06/ft6x06.o: In function `ft6x06_TS_ITStatus':
ft6x06.c:(.text.ft6x06_TS_ITStatus+0x0): multiple definition of `ft6x06_TS_ITStatus'
BUILD/DISCO_F413ZH/GCC_ARM-RELEASE/BSP_DISCO_F413ZH/Drivers/BSP/Components/ft6x06/ft6x06.o:ft6x06.c:(.text.ft6x06_TS_ITStatus+0x0): first defined here
...
BUILD/DISCO_F413ZH/GCC_ARM-RELEASE/F413ZH_SD_BlockDevice/BSP_DISCO_F413ZH/Utilities/Fonts/font8.o:(.rodata.Font8_Table+0x0): multiple definition of `Font8_Table'
BUILD/DISCO_F413ZH/GCC_ARM-RELEASE/BSP_DISCO_F413ZH/Utilities/Fonts/font8.o:(.rodata.Font8_Table+0x0): first defined here
collect2: error: ld returned 1 exit status

[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/home/drenyovszki_rajmund/utensor-mnist-demo"
       Command: "/usr/bin/python -u /home/drenyovszki_rajmund/utensor-mnist-demo/mbed-os/tools/make.py -t GCC_ARM -m DISCO_F413ZH --profile uTensor/build_profile/release.json --source . --build ./BUILD/DISCO_F413ZH/GCC_ARM-RELEASE -c"
       Tip: You could retry the last command with "-v" flag for verbose output
---

neil-tan, 2018-11-12:
Hey @drajmund 
  
You shouldn't need to manually add the BSP_DISCO_F413ZH. Is there an error/warning led you to do so?

It would be useful to see the error after this line:
`(.text.ft6x06_TS_ITStatus+0x0): first defined here`
Also, the output of this comment at your current project root:
`tree -L 4 -I '*.idx|mbed-os'`

In the meanwhile, can we try a fresh import of the project? Here are the commands I used to get it going:

```
mbed import https://github.com/uTensor/utensor-mnist-demo
cd utensor-mnist-demo/
mbed compile -m DISCO_F413ZH -t GCC_ARM --profile=uTensor/build_profile/release.json
```

To use the SD-free version:

```
mbed import https://github.com/uTensor/utensor-mnist-demo
cd utensor-mnist-demo/
git checkout f/rom-tensor
mbed deploy
mbed compile -m DISCO_F413ZH -t GCC_ARM --profile=uTensor/build_profile/release.json -c
```

drajmund, 2018-11-12:
Oh, the problem was exactly what you mentioned: I manually added BSP_DISCO_F413ZH. (Root cause may be the copy of BSP_DISCO_F413ZH.lib two times?)
After imorting the project and compiling it again I received this :-)
...
Compile [100.0%]: test_env.cpp
Link: utensor-mnist-demo
Elf2Bin: utensor-mnist-demo
...
Total Static RAM memory (data + bss): 19712 bytes
Total Flash memory (text + data): 357041 bytes
Image: ./BUILD/DISCO_F413ZH/GCC_ARM-RELEASE/utensor-mnist-demo.bin

Thank you for your help. I think I have learned a lot from this.
The tree command output was (before reimporting the project):
.
├── BSP_DISCO_F413ZH
│   ├── Drivers
│   │   └── BSP
│   │       ├── Components
│   │       └── STM32F413H-Discovery
│   └── Utilities
│       └── Fonts
│           ├── font12.c
│           ├── font16.c
│           ├── font20.c
│           ├── font24.c
│           ├── font8.c
│           └── fonts.h
├── BSP_DISCO_F413ZH.lib
├── BUILD
│   └── DISCO_F413ZH
│       ├── ARM-RELEASE
│       │   ├── BSP_DISCO_F413ZH
│       │   ├── F413ZH_SD_BlockDevice
│       │   ├── mbed_config.h
│       │   ├── models
│       │   └── uTensor
│       └── GCC_ARM-RELEASE
│           ├── BSP_DISCO_F413ZH
│           ├── F413ZH_SD_BlockDevice
│           ├── main.d
│           ├── main.o
│           ├── mbed_config.h
│           ├── models
│           ├── uTensor
│           └── utensor-mnist-demo.map
├── constants
│   └── deep_mlp
├── docs
│   └── images
│       ├── jupyter.png
│       ├── kernel.png
│       └── train_mlp.png
├── F413ZH_SD_BlockDevice
│   ├── BSP_DISCO_F413ZH
│   │   ├── Drivers
│   │   │   └── BSP
│   │   └── Utilities
│   │       └── Fonts
│   ├── BSP_DISCO_F413ZH.lib
│   ├── F413ZH_SD_BlockDevice.cpp
│   ├── F413ZH_SD_BlockDevice.h
│   └── README.md
├── F413ZH_SD_BlockDevice.lib
├── image.h
├── LICENSE
├── main.cpp
├── mbed_app.json
├── mbed-os.lib
├── mbed_settings.py
├── mbed_settings.pyc
├── models
│   ├── deep_mlp.cpp
│   ├── deep_mlp.hpp
│   └── deep_mlp_weight.hpp
├── README.md
├── requirements.txt
├── simconfig.json
├── tensorflow-models
│   ├── deep_mlp.ipynb
│   └── deep_mlp.py
├── uTensor
│   │   └── release.json
│   ├── CMakeLists.txt
│   ├── contribution_guide.md
│   ├── deprecated
│   │   └── TESTS
│   │       ├── ArrayTests.hpp
│   │       ├── context_test.hpp
│   │       ├── main.cpp
│   │       ├── MathTests.hpp
│   │       ├── MatrixTests.hpp
│   │       ├── mlp_test.hpp
│   │       ├── NnTests.hpp
│   │       ├── scripts
│   │       ├── sdtest.hpp
│   │       ├── tensorIdxImporterTests.hpp
│   │       ├── TensorImport
│   │       ├── tensor_test.hpp
│   │       ├── test.cpp
│   │       ├── test.hpp
│   │       └── vmtest.hpp
│   ├── docs
│   │   ├── img
│   │   │   └── uTensorFlow.jpg
│   │   └── newProject.md
│   ├── examples
│   │   ├── deep_mnist_mlp.cpp
│   │   └── deep_mnist_mlp.hpp
│   ├── LICENSE
│   ├── mbed_app.json
│   ├── mbed_lib.json
│   ├── README.md
│   ├── reference
│   │   ├── models
│   │   │   ├── miscOps
│   │   │   └── qntMatMulDeqnt
│   │   ├── README.md
│   │   └── tf_src
│   │       ├── CXX
│   │       └── python
│   ├── REFERENCES.md
│   ├── requirements.txt
│   ├── TESTS
│   │   ├── constants
│   │   │   ├── ArgMax
│   │   │   ├── deep_mlp
│   │   │   ├── deQ
│   │   │   ├── idxImport
│   │   │   ├── import-MatMul_eightbit_requantize
│   │   │   ├── mlpTest
│   │   │   ├── qA
│   │   │   ├── qB
│   │   │   ├── qMatMul
│   │   │   ├── ref_add
│   │   │   ├── ref_max
│   │   │   ├── ref_min
│   │   │   ├── ref_qRelu
│   │   │   ├── ref_reshape
│   │   │   ├── rQ
│   │   │   ├── rqRange
│   │   │   ├── sdtmp
│   │   │   ├── test
│   │   │   └── test_quantized_add
│   │   ├── core
│   │   │   ├── context
│   │   │   ├── tensor
│   │   │   ├── tensorIdxImporter
│   │   │   └── virtualmem
│   │   ├── dummy
│   │   │   └── sanity
│   │   ├── gtest_main.cpp
│   │   ├── operators
│   │   │   ├── arrayOps
│   │   │   ├── basicMath
│   │   │   ├── matrix
│   │   │   └── quantizedAdd
│   │   ├── README.md
│   │   └── test_helper.h
│   ├── unity
│   │   └── CMakeLists.txt
│   └── uTensor
│       ├── core
│       │   ├── context.cpp
│       │   ├── context.hpp
│       │   ├── sdtensor.hpp
│       │   ├── tensor.cpp
│       │   ├── tensor.hpp
│       │   ├── uTensorBase.cpp
│       │   ├── uTensorBase.hpp
│       │   ├── vm.cpp
│       │   └── vm.hpp
│       ├── loaders
│       │   ├── tensorIdxImporter.cpp
│       │   └── tensorIdxImporter.hpp
│       ├── ops
│       │   ├── ArrayOps.hpp
│       │   ├── MathOps.hpp
│       │   ├── MatrixOps.hpp
│       │   └── NnOps.hpp
│       └── util
│           ├── math_utils.hpp
│           ├── quantization_utils.cpp
│           ├── quantization_utils.hpp
│           ├── uTensor_util.cpp
│           └── uTensor_util.hpp
└── uTensor.lib