uTensor/uTensor #139: Error when compiling newProject

Akselvb, 2019-01-14:
Hello again!

This is more of a question rather than issue. I am trying to do the newProject tutorial on my NRF52_DK, but I get an mbed compile error ('region FLASH overflowed by ... bytes' and '.text will not fit in region FLASH'). For me it doesn't make any sense, because I've changed the size of the NN such that the trained weights file should fit onto the Flash.

I wish to try to compile it using the mbed online compiler, to see if it makes any difference. However, the total size of the folder is ~850MB. I am therefore wondering whether I can leave out some (unnecessary) files before uploading it. Or if anyone knows of any workarounds using the mbed cli, that would also be much appreciated.

I can add that I am certainly no expert when it comes to embedded programming.

Thanks.

mbartling, 2019-01-14:
Hi @Akselvb,

Do you have any additional programs/c files in your project. For example a Python Virtual env? The online compiler/mbed build tools automatically locate all c/cpp style files to build. Also can you post the image from the build tab on the righthand side?
Should look something like this:
[image]

I'm curious how your build is coming out to 850 MB, that's definitely too big to fit on these boards. Mbed-OS repo is 200+ MB, but 99.99999% gets thrown away during the linking step. How big are the weight tensors in your project?

Akselvb, 2019-01-14:
Hi @mbartling 

I am following the exact steps of the tutorial, so no additional files. I am doing it with a Python Virtual env (2.7.15) activated, but it doesn't look like any additional files are generated for that reason. The _mbed-os/_ folder generated by the 'mbed new' command is 825MB, is that normal?

[image]

I haven't been able to upload the project to the online compiler yet, due to the size.

The total size of the _models/_ folder including the tensor weights is not more than 74kB, so it shouldn't be a problem.

mbartling, 2019-01-14:
cool, that was actually my main question. Yeah mbed os has a bunch of platform specific stuffs and that takes up a massive amount of space. The build tools know what to compile and ignore + link, so mbed-os will take up approximately 20kB after build. 

Just double checking, but the python venv is not in your project directory right? It has a bunch of C files that may either break the build or make it HUGE. sometimes it shows up as a hidden folder in the project root. 

I think I figured out the error and it is silly. Basically, the demo relies on the old release which defaults to SD card stuff, but the CLI defaults to storing the weights in ROM (@dboyliao). @neil-tan The interface in main is broken, but that's a separate issue. The primary reason you arent able to compile is because the SD driver upstream is broken right now. 

A patch to get this working: remove the `sd-driver.lib` and `sd-driver` directory. Then `cd utensor &&  git checkout develop && cd ..`. Develop has the ROM tensor code enabled. 

Now main needs to change a bit to handle the switch from SD tensors to ROM Tensors (BinaryTensor).
@Knight-X Do we have a TensorImporter interface for the ROM tensors? @Akselvb I can help update this main file, just need a bit of time.

Knight-X, 2019-01-15:
ROM tensors have no TensorImporter interface. It should be prepared with codegen. @Akselvb  What do you mean " update image"?

Akselvb, 2019-01-15:
@Knight-X I mean when compiling using mbed cli, I get the following result:

[image]

It should be _Image: hello_world.bin_ instead of _Update Image: hello_world_update.bin_, right?

neil-tan, 2019-01-15:
My guess is `hello_world.hex`. This is unusually, as I'd expect to see an `.bin` extension as oppose to `.hex`, normally.

mbartling, 2019-01-15:
Update image is an extension for mbed Cloud Client, which shouldn't be built unless mbed-cloud-client is part of the project. This is weird behavior for mbed-cli. I don't think it's actually related to uTensor.

Note: Taken from the board page on [os.mbed.com](https://os.mbed.com/platforms/Nordic-nRF52-DK/)
```compiled binary size
For the default DK mbed target, any mbed build tools (online IDE, mbed CLI) will put the SoftDevice and the user application together into a binary file intended to program your development kit. Be aware of the SoftDevice size – it adds more than a hundred KB to any compiled binaries.

Further information about the type of Nordic’s target is available in nRF5 ReadMe.
```

We may have to forward this question to github.com/armmbed/mbed-os as it might be out of scope. @neil-tan What do you think?

Akselvb, 2019-01-18:
Unfortunately, even with the smallest possible NN architecture (1 single neuron), I get the same error message when compiling with NRF52_DK targeted. It is probably me doing something wrong. Can anyone try to compile with the same target and see if they get the same error message? :-)

neil-tan, 2019-01-18:
@Akselvb Same error here. I will be flying to Europe in 7 hours and won't be able to look at this further. It would be interesting to look how big of each part of the program is. Perhaps, using the compilation summary for a different target? Will ping some of the Arm folks, see if they have some quick answers.

neil-tan, 2019-01-18:
For K66F:
```
| Module             |           .text |       .data |          .bss |
|--------------------|-----------------|-------------|---------------|
| [fill]             |       505(+505) |     11(+11) |       28(+28) |
| [lib]/c.a          |   69431(+69431) | 2548(+2548) |     127(+127) |
| [lib]/gcc.a        |     7456(+7456) |       0(+0) |         0(+0) |
| [lib]/m.a          |       788(+788) |       0(+0) |         0(+0) |
| [lib]/misc         |       248(+248) |       8(+8) |       28(+28) |
| [lib]/nosys.a      |         32(+32) |       0(+0) |         0(+0) |
| [lib]/stdc++.a     | 173167(+173167) |   141(+141) |   5676(+5676) |
| main.o             |     1101(+1101) |       0(+0) |     105(+105) |
| mbed-os/cmsis      |     1033(+1033) |       0(+0) |         0(+0) |
| mbed-os/components |       120(+120) |       0(+0) |         0(+0) |
| mbed-os/drivers    |     1288(+1288) |       0(+0) |         0(+0) |
| mbed-os/features   |       172(+172) |       0(+0) |     192(+192) |
| mbed-os/hal        |     1614(+1614) |       4(+4) |       68(+68) |
| mbed-os/platform   |     3775(+3775) |   260(+260) |     246(+246) |
| mbed-os/rtos       |     7231(+7231) |   168(+168) |   5973(+5973) |
| mbed-os/targets    |     7061(+7061) |     12(+12) |     301(+301) |
| models/deep_mlp.o  |   37006(+37006) |       0(+0) |         1(+1) |
| uTensor/uTensor    |     6851(+6851) |       0(+0) |         7(+7) |
| Subtotals          | 318879(+318879) | 3152(+3152) | 12752(+12752) |
```

neil-tan, 2019-01-19:
@Akselvb Ok, according to my colleague, the NRF52_DK will require an additional 100kb+ for the BLE stack. Given the module breakdown above, it's not going to fit.

However, we are working on a change that will reduce the dependency on stdc++, which, should cut its binary size significantly. @mbartling

neil-tan, 2019-01-20:
@Akselvb 
Send us an email at utensor@googlegroups.com
I can invite you to our slack channel, if that's of interest.

closed 2019-01-20 by neil-tan