uTensor/uTensor #141: How to add more operator?

auvilink, 2019-02-02:
Hi, guys, great job so far.

I tried to export TF trained model (simple audio recognition) to utensor, however, there are many operators in my pb file not supported by utensor, for example:
DecodeWav, AudioSpectrogram, Mfcc, Identity, Conv2D, BiasAdd...

Any chance I can add those operators to uTensor? If so, how?

Thanks.

dboyliao, 2019-02-03:
Hi, @auvilink.
It involves two things:
1. the `uTensor` runtime kernel
2. code generator support

For `1`, let's take `Relu` as example.
You can take a look over [here](https://github.com/uTensor/uTensor/blob/master/uTensor/ops/NnOps.hpp).
A `Relu` consist of two parts, the kernel and an Op using that kernel.
The template function,`Relu`, is the kernel and `ReluOp` is the operator.
The op interface is defined over [here](https://github.com/uTensor/uTensor/blob/master/uTensor/core/uTensorBase.hpp).
So you can extend the kernel following this approach.

After the C++ kernel is done, you also need to add it to the code generator.
I recommend you to work with `develop` branch of `utensor_cgen`.
You can find it [here](https://github.com/uTensor/utensor_cgen/tree/develop).
To be specific, you need to take a look of its `backend`.
Basically, you need to add template for the snippet objects (you can find it over `backend/snippet` submodule) and the operator object (`backend/operators.py`).
I'm sorry that this part is kinda messy for now.
Let me know if you have any issue about the code generator part.

Of course, if you want to write C++ by hands, `2` is not an issue.

dboyliao, 2019-02-03:
Or you can make a PR to `uTensor` for the c++ kernel, and then my teammate and I will add them to the code generator.
Maybe that'd be faster.

auvilink, 2019-02-03:
@dboyliao 
Thanks for the explanation.

I'll try to figure it out.