uTensor/uTensor #151: ReluOp - template brocken since 4cb09cb8b4d0859405e30f64087b76085d227bd3

miki89, 2019-04-14:
Example from https://blog.hackster.io/simple-neural-network-on-mcus-a7cbd3dc108c does not compile any longer:
due to error: 

> .\models\deep_mlp.cpp:182:48: error: wrong number of template arguments (3, should be 2)
     ctx.push(new ReluOp<uint8_t, float, uint8_t>(),
                                                ^
In file included from .\models\deep_mlp.cpp:8:
./uTensor/uTensor/ops/NnOps.hpp:76:7: note: provided for 'template<class T1, class TOut> class ReluOp'
 class ReluOp : public Operator {
       ^~~~~~
.\models\deep_mlp.cpp:307:48: error: wrong number of template arguments (3, should be 2)
     ctx.push(new ReluOp<uint8_t, float, uint8_t>(),
                                                ^
In file included from .\models\deep_mlp.cpp:8:
./uTensor/uTensor/ops/NnOps.hpp:76:7: note: provided for 'template<class T1, class TOut> class ReluOp'
 class ReluOp : public Operator {
       ^~~~~~

caused, by commit of jakubvalis88 on 6 Mar commit 4cb09cb8b4d0859405e30f64087b76085d227bd3

previous commit (1c8eca4a7bbd952a87e5e9b2d055ba4131d16a3a) compiles. 

Please have a look.

mbartling, 2019-04-14:
Hi, I actually have a PR for this open in codegen. 

https://github.com/uTensor/utensor_cgen/pull/64

miki89, 2019-04-14:
Hi, 
sure I understand that, but what confused me also it that the TESTS, even if deprecated and the examples show 3 arguments instead of 2. 
https://github.com/uTensor/uTensor/search?q=ReluOp&unscoped_q=ReluOp 

Do you also have more a stable branch, where the releases bewteen cgen and uTensor core are synced with each other? 
using the develop branch might not always be the best choice. 

Thank you again for commenting!

mbartling, 2019-04-14:
Yes, we use master as stable :)

neil-tan, 2019-04-15:
@miki89 I agree we should make the working combination more obvious. Perhaps, we should include the tags in the Readme?

mbartling, 2019-05-28:
The issue is we migrated the old RELU op (which was quantized) to one with "quantized" in the name for clarity.