uTensor/uTensor #155: "Creating a New Project" tutorial is broken

alejoseb, 2019-05-08:
Hi,
I was testing the examples provided, the first one "Creating a New Project" is not working currently.
It is not compiling with the current version of mbed ( mbed-os-5.12.3). It seems that the build profile to enable C++11 support is not compatible with mbed mbed_rtc_time.cpp file.
The other two examples works as expected.

is it possible to update the example or avoid the C++11 profile?

Part of the output is as follow:

Compile [ 88.4%]: mbed_rtc_time.cpp
[Error] mbed_rtc_time.cpp@89,22: invalid use of incomplete type 'const struct timeval'
[Error] mbed_rtc_time.cpp@110,7: invalid use of incomplete type 'struct timeval'
[Error] mbed_rtc_time.cpp@111,7: invalid use of incomplete type 'struct timeval'
[Error] mbed_rtc_time.cpp@124,20: aggregate 'timeval tv' has incomplete type and cannot be defined
[Error] mbed_rtc_time.cpp@137,26: variable 'const timeval tv' has initializer but incomplete type
[ERROR] ./mbed-os/platform/mbed_rtc_time.cpp: In function 'int settimeofday(const timeval*, const timezone*)':
./mbed-os/platform/mbed_rtc_time.cpp:89:22: error: invalid use of incomplete type 'const struct timeval'

Thanks for your help

mbartling, 2019-05-08:
Hey @alejoseb,

Thanks for letting us know, completely forgot about this change in the latest couple of mbed-os releases. Give me a few min, will try to fix it now

-Michael

mbartling, 2019-05-08:
Update, can confirm this is actually an issue with mbed OS.

Going to raise an issue with them now.

mbartling, 2019-05-08:
https://github.com/ARMmbed/mbed-os/issues/10548
https://github.com/ARMmbed/mbed-os/issues/10547

neil-tan, 2019-05-09:
@alejoseb.  The temporary fix would be:
```
cd mbed-os
git checkout mbed-os-5.11
mbed sync
cd ..
```
This should rollback to `mbed-os-5.11`

kjbracey-arm, 2019-05-21:
The issue is that your build profile pushes the language version up to C++11, but disables GNU extensions, because it's selecting `-std=c++11` not `-std=gnu++11`.

Mbed OS should work fine built as C++11, but I wouldn't expect it to be happy with GNU extensions disabled.

kjbracey-arm, 2019-05-24:
Both mbed os issues linked above have been closed (as "not a bug"), so this should probably be re-opened.

mbartling, 2019-05-24:
@kjbracey-arm the uTensor codebase was designed around vanilla C++11 for portability. Do you see any issues with enabling gnu extensions limiting the scope of uptake? If not I have no problem updating the build profile

kjbracey-arm, 2019-05-24:
I'm not really in favour of using GNU extensions either, but the Mbed OS build profiles have always had them on for GCC and ARM Compiler, so actually turning them off is liable to upset the Mbed OS half of the build. The GNU extensions don't *normally* actually interfere with standard C++11 code.

I'd suggest you could use a `-std=c++11` build to test your codebase (maybe in a non Mbed-OS application build?), but not for a full Mbed OS build, where you have to choose one language standard for the complete build.

Also, if you're publishing a variant Mbed OS build profile, you're liable to hit glitches with new Mbed OS versions - we occasionally do have to make changes to them.

If https://github.com/ARMmbed/mbed-os/pull/10427 lands then you wouldn't actually need to use a profile for Mbed OS 5.13 or later, as it will be selecting (GNU) C++14, so should compile your code fine.

mbartling, 2019-05-24:
I guess the build profiles are mbed specific anyways so there really is not harm updating them to the gnu extensions for the time being.

karimkhanvi, 2019-08-09:
> @alejoseb. The temporary fix would be:
> 
> ```
> cd mbed-os
> git checkout mbed-os-5.11
> mbed sync
> cd ..
> mbed deploy
> ```
> 
> This should rollback to `mbed-os-5.11`

Thanks this is good quick fix.