uTensor/uTensor #160: TensorFlow outputs vector, uTensor outputs single value

janjongboom, 2019-06-20:
Maybe (perhaps!) I don't completely understand how uTensor's mapping works, but I have a neural network that has 3 output neurons (Keras code):

```py
model = Sequential()
model.add(Dense(40, input_dim=186, activation='relu', name='x')) 
model.add(Dropout(0.5, seed=5, name='dropout1'))
model.add(Dense(25, activation='relu', name='hidden3'))
model.add(Dense(5, activation='relu', name='hidden5'))
model.add(Dense(3, activation='softmax', name='y_pred'))
```

When I run a classification in TensorFlow I get a three-dimensional vector back, e.g.:

```
[9.9981266e-01, 1.8708695e-04, 2.8274363e-07]
```

However, uTensor seems to think it should output a single value, rather than a vector. utensor-codegen outputs:

```cpp
//...

{   
    ctx.add(new RamTensor<uint8_t>(), "y_pred/BiasAdd/eightbit/requantize:0", 1);
    ctx.add(new RamTensor<float>({1}), "y_pred/BiasAdd/eightbit/requantize:1", 1);
    ctx.add(new RamTensor<float>({1}), "y_pred/BiasAdd/eightbit/requantize:2", 1);
    ctx.push(new RequantizeOp(),
             { "y_pred/BiasAdd/eightbit:0", "y_pred/BiasAdd/eightbit:1", "y_pred/BiasAdd/eightbit:2", "y_pred/BiasAdd/eightbit/requant_range:0", "y_pred/BiasAdd/eightbit/requant_range:1" },
             { "y_pred/BiasAdd/eightbit/requantize:0", "y_pred/BiasAdd/eightbit/requantize:1", "y_pred/BiasAdd/eightbit/requantize:2" });
    ctx.eval();
}
{
    ctx.add(new RamTensor<float>(), "y_pred/BiasAdd:0", 1);
    ctx.push(new DequantizeOp(), 
             { "y_pred/BiasAdd/eightbit/requantize:0", "y_pred/BiasAdd/eightbit/requantize:1", "y_pred/BiasAdd/eightbit/requantize:2" },
             { "y_pred/BiasAdd:0" });
    ctx.eval();
}
{
    ctx.add(new RamTensor<float>(), "y_pred/Softmax:0");
    ctx.push(new SoftmaxOp(),
             { "y_pred/BiasAdd:0" },
             { "y_pred/Softmax:0" });
}
```

This looks like a single value to me.

```
ctx.get("y_pred/Softmax:0")
```

Seems to also return a single float.

This is the output for `utensor-cli show --oneline`:

```
x_input op_type: Placeholder, inputs: [], outputs: ['x_input:0']
x/kernel op_type: Const, inputs: [], outputs: ['x/kernel:0']
x/kernel/read op_type: Identity, inputs: ['x/kernel:0'], outputs: ['x/kernel/read:0']
x/bias op_type: Const, inputs: [], outputs: ['x/bias:0']
x/bias/read op_type: Identity, inputs: ['x/bias:0'], outputs: ['x/bias/read:0']
x/MatMul op_type: MatMul, inputs: ['x_input:0', 'x/kernel/read:0'], outputs: ['x/MatMul:0']
x/BiasAdd op_type: BiasAdd, inputs: ['x/MatMul:0', 'x/bias/read:0'], outputs: ['x/BiasAdd:0']
x/Relu op_type: Relu, inputs: ['x/BiasAdd:0'], outputs: ['x/Relu:0']
dropout1/keras_learning_phase/input op_type: Const, inputs: [], outputs: ['dropout1/keras_learning_phase/input:0']
dropout1/keras_learning_phase op_type: PlaceholderWithDefault, inputs: ['dropout1/keras_learning_phase/input:0'], outputs: ['dropout1/keras_learning_phase:0']
dropout1/cond/Switch op_type: Switch, inputs: ['dropout1/keras_learning_phase:0', 'dropout1/keras_learning_phase:0'], outputs: ['dropout1/cond/Switch:0', 'dropout1/cond/Switch:1']
dropout1/cond/switch_t op_type: Identity, inputs: ['dropout1/cond/Switch:1'], outputs: ['dropout1/cond/switch_t:0']
dropout1/cond/pred_id op_type: Identity, inputs: ['dropout1/keras_learning_phase:0'], outputs: ['dropout1/cond/pred_id:0']
[... 48 lines cut ...]
training/Adam/Variable_1 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_1:0']
training/Adam/Variable_2 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_2:0']
training/Adam/Variable_3 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_3:0']
training/Adam/Variable_4 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_4:0']
training/Adam/Variable_5 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_5:0']
training/Adam/Variable_6 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_6:0']
training/Adam/Variable_7 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_7:0']
training/Adam/Variable_8 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_8:0']
training/Adam/Variable_9 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_9:0']
training/Adam/Variable_10 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_10:0']
training/Adam/Variable_11 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_11:0']
training/Adam/Variable_12 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_12:0']
training/Adam/Variable_13 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_13:0']
training/Adam/Variable_14 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_14:0']
training/Adam/Variable_15 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_15:0']
training/Adam/Variable_16 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_16:0']
training/Adam/Variable_17 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_17:0']
training/Adam/Variable_18 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_18:0']
training/Adam/Variable_19 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_19:0']
training/Adam/Variable_20 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_20:0']
training/Adam/Variable_21 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_21:0']
training/Adam/Variable_22 op_type: Const, inputs: [], outputs: ['training/Adam/Variable_22:0']
```

I've put the whole project here for reference: https://github.com/janjongboom/utensor-test

dboyliao, 2019-06-20:
I think `y_pred/Softmax:0` should be of shape `1x3`.
You can use `Tensor::getShape` to see what the shape is:
https://github.com/uTensor/uTensor/blob/develop/uTensor/core/tensor.cpp#L99

Here is the declaration of `TensorShape`:
https://github.com/uTensor/uTensor/blob/develop/uTensor/core/tensor.hpp#L30-L34

janjongboom, 2019-06-21:
@dboyliao @neil-tan  This returns `2` ?!

```
    S_TENSOR pred_tensor = ctx.get("y_pred/Softmax:0");  // getting a reference to the output tensor
    printf("Size: %lu\n", pred_tensor->getShape().size());
```

The plot thickens...

Knight-X, 2019-06-21:
Weird. It should be 3.

janjongboom, 2019-06-21:
@Knight-X @dboyliao Side-question: How am I supposed to loop over this?

```
    printf("Predicted label 1: %f\r\n", *(pred_tensor->read<float>(0, 0)));
    printf("Predicted label 2: %f\r\n", *(pred_tensor->read<float>(4, 0)));
    printf("Predicted label 3: %f\r\n", *(pred_tensor->read<float>(8, 0)));
```

?

dboyliao, 2019-06-21:
```
float* ptr_pred = pred_tensor->read<float>(0, 0);
printf("%f\r\n", *ptr_pred);
printf("%f\r\n", *(ptr_pred + 1));
printf("%f\r\n", *(ptr_pred + 2));
```

That's how I do it.

neil-tan, 2019-06-21:
`printf("Size: %lu\n", pred_tensor->getShape().size());` gives you the number of dimensions of the tensor.   1x3 has 2 dimensions

janjongboom, 2019-06-22:
Figured it out with @neil-tan, but it'd be great if the codegen can handle this for me, instead of just randomly casting values.

closed 2019-06-22 by janjongboom

dboyliao, 2019-06-22:
What do you mean by randomly casting?

janjongboom, 2019-06-22:
@dboyliao Codegen knows the size and the type of my output layer. Thus retrieving that tensor should have the right type and size already available. Not just return `const void*` to some piece of memory and let me handle the casting to the right type and making sure I iterate over the memory correctly.

Same with input layer actually. No checks on type and size of what you pass in.

Knight-X, 2019-06-22:
float* ptr_pred = pred_tensor->read<float>(0, 0);
@janjongboom do you mean that cast from void* to typed ptr ？ I think that would be a issue of runtime design. Or you mean that we should generate the inference code instead of writing manually?