uTensor/uTensor #179: circleci build fails: Not a git repository

janjongboom, 2019-09-03:
@mbartling 

When running `circleci build` against `revert-172-revert-171-fix-softmax` branch:

```
====>> Checkout code
  #!/bin/bash -eo pipefail
mkdir -p /root/project && cd /tmp/_circleci_local_build_repo && git ls-files | tar -T - -c | tar -x -C /root/project && cp -a /tmp/_circleci_local_build_repo/.git /root/project
fatal: Not a git repository: ../.git/modules/uTensor
Error: Exited with code 128
Step failed
====>> Uploading test results
Error: Unable to save test results from /tmp/test_results
Error stat /tmp/test_results: no such file or directory

Error: Found no path with test results, skipping
====>> Uploading artifacts
Uploading /tmp/coverage to tmp/coverage
  No artifact files found at /tmp/coverage
Error: runner failed (exited with 101)
Task failed
Error: task failed
```

dboyliao, 2019-09-03:
Have you tried `git submodule update --init --recursive`?

dboyliao, 2019-09-03:
Looks like you forget to initialize the submodules

janjongboom, 2019-09-03:
Ah, OK. It built fine otherwise, so hadn't realized that.

closed 2019-09-03 by janjongboom