uTensor/uTensor #182: Tensor setup via method chaining

dboyliao, 2019-10-01:
Sth like this:
```
Tensor* t = new Tensor();
(t->setShape({3, 3})->setAddress(<address>))....
```
It'll be a lot easier to design a template for the code generator if we have many mechanism to control how tensor behave at the runtime.
Mostly mem allocation for now, but I can see there'll be more to come in the future.

mbartling, 2019-10-03:
I like the idea of chaining, but what do you mean by setAddress?

dboyliao, 2019-10-04:
Oh, It's about @Knight-X's mem allocation optimization pass.
That is, allocate the tensor at specific address (in mem pool managed by uTensor runtime).

neil-tan, 2019-10-04:
aka. offline memory planning

dboyliao, 2019-10-04:
The naming is just for illustrate the idea.
Maybe we can use another more informative name.

mbartling, 2019-10-15:
Force addressing here is kinda scary, especially if you rope in linker scripts like most people do in production environments

mbartling, 2019-10-15:
Method chaining can also be done super efficiently :D