uTensor/uTensor #184: State of convolution?

janjongboom, 2019-11-10:
Hi @neil-tan, @dboyliao,

What's the state of convolutional neural networks? I thought they were supported already... This is my Keras model definition:

```
from keras.models import Sequential
from keras.layers import Dense, InputLayer, Dropout, Conv2D, Flatten, Reshape, MaxPooling2D
from keras.optimizers import Adam

model = Sequential()
model.add(Reshape((125, 3, 1) ,input_shape=(X_train.shape[1], ), name='x_input'))
model.add(Conv2D(20, kernel_size=1, activation='relu'))
model.add(MaxPooling2D(pool_size=1))
model.add(Conv2D(10, kernel_size=1, activation='relu'))
model.add(MaxPooling2D(pool_size=1))
model.add(Flatten())
model.add(Dense(classes, activation='softmax', name='y_pred'))

opt = Adam(lr=0.0001, beta_1=0.9, beta_2=0.999)

model.compile(loss='categorical_crossentropy', optimizer=opt, metrics=['accuracy'])
model.fit(X_train, Y_train, batch_size=50, epochs=100, validation_data=(X_test, Y_test))
```

Which yields the following error when exporting to uTensor (quantization disabled):

```
[INFO code_generator.py _generate @ 71] Transforming graph: user/1/learn-289/trained.pb
[INFO code_generator.py _generate @ 72] Transform pipeline: dropout -> inline -> biasAdd -> remove_id_op -> refcnt
[INFO code_generator.py _generate @ 76] Graph transormation done
Traceback (most recent call last):
  File "/usr/local/bin/utensor-cli", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/utensor_cgen/cli.py", line 84, in convert_graph
    generator.generate(model_path)
  File "/usr/local/lib/python3.7/site-packages/utensor_cgen/backend/code_generator.py", line 51, in generate
    self._generate(src_fname, ugraph)
  File "/usr/local/lib/python3.7/site-packages/utensor_cgen/backend/code_generator.py", line 103, in _generate
    weight_container=weight_container)
  File "/usr/local/lib/python3.7/site-packages/utensor_cgen/backend/operators.py", line 22, in createOperatorSnippet
    raise ValueError(err_msg)
ValueError: unsupported op type in uTensor: Prod
```

Not sure where `Prod` actually comes from.

janjongboom, 2019-11-10:
`AvgPool` also doesn't work, so I'm wondering which pooling layers are supported...

dboyliao, 2019-12-04:
As the error message suggests, there is an unsupported op, `Prod`, in your graph.
Not sure if its about how Keras implements conv layer.
I'm pretty sure plain TF conv is supported in uTensor.

Pooling layers are defined here:
https://github.com/uTensor/uTensor/blob/develop/src/uTensor/ops/NnOps.hpp
And yes, we have only max pooling.

closed 2020-10-19 by mbartling