uTensor/uTensor #215: Overflow on TensorStrides

victorromeo, 2020-11-17:
The TensorStrides constructor implementation in 976ad844950ed1c683e8b8e1a7c42ed8a0a62f61 overflows, due to use of unsigned type in the second for loop, caused by decrementing below 0.
```
TensorStrides::TensorStrides(TensorShape& shape) {
  _num_dims = shape.num_dims();
  size_t last_idx = _num_dims - 1;
  for (size_t i = last_idx + 1; i < 3; ++i) {
    _strides[i] = 0;
  }
  _strides[last_idx] = 1;
  uint32_t s = 1;
  for (size_t i = last_idx - 1; i >= 0; --i) {
    s *= shape[i + 1];
    _strides[i] = s;
  }
}
```
Recommend signed indexor of type int32_t
```
TensorStrides::TensorStrides(TensorShape& shape) {
  _num_dims = shape.num_dims();
  size_t last_idx = _num_dims - 1;
  for (size_t i = last_idx + 1; i < 3; ++i) {
    _strides[i] = 0;
  }
  _strides[last_idx] = 1;
  uint32_t s = 1;
  for (int32_t i = last_idx - 1; i >= 0; --i) {
    s *= shape[i + 1];
    _strides[i] = s;
  }
}
```

dboyliao, 2020-11-17:
@victorromeo Thanks for reporting this issue.
Could you please give me an example which may cause the issue you report?
I don't quite get why the `i` in the second loop will go below 0.

victorromeo, 2020-11-17:
As a minor correction, the issue is underflow (not overflow), caused by the assessment of `i>=0` which repeats after `--i' when `i ==0`.  As a result the calculation of the internal loop operations loop can loop indefinitely.  Will provide example.

Revert to use `size_t` instead of `int32_t` and run the unit tests checked in under #216 a segmentation fault occurs, due to an endless loop exhausting.

```
332/341 Test #332: QuantTanhTest.sq_tanh_4 ..................................   Passed    0.00 sec
        Start 333: Transpose.transpose_test
333/341 Test #333: Transpose.transpose_test .................................***Exception: SegFault  0.13 sec
        Start 334: TopInclude.AddOp
334/341 Test #334: TopInclude.AddOp .........................................   Passed    0.00 sec
        Start 335: Model.example1
```

victorromeo, 2020-11-17:
An alternative fix would be to continue using size_t but increment the indexor used by the for loop.

```c++
TensorStrides::TensorStrides(TensorShape& shape) {
  _num_dims = shape.num_dims();
  size_t last_idx = _num_dims - 1;
  for (size_t i = last_idx + 1; i < 3; ++i) {
    _strides[i] = 0;
  }
  _strides[last_idx] = 1;
  uint32_t s = 1;
  for (size_t i = last_idx; i > 0; --i) {
    s *= shape[i];
    _strides[i + 1] = s;
  }
}
```

dboyliao, 2020-11-17:
@victorromeo Nice catch.
I didn't notice `>= 0` will cause such underflow.
I don't under stand why your alternative fix works.
`_strides[i+1] = s;` looks weird to me.

Let's say we have 3 dim tensor with shape [3, 4, 2].
The expecting `_strides` should be `[8, 2, 1, 0]`.
In the first loop, the `_strides` is initialized to `[*, *, *, 0]`
And the `last_idx` is `2`.

In the second loop, `s = 1`:
`i = 2`: `s *= shape[2] # which is 2` -> `_strides[3] = 2 // which makes _strides = [*, *, *, 2]`
And follow the loop the `_strides` seems not correctly initialized.

As a result, I think using `int32_t` is a better fix to me.

balajikulkarni, 2020-12-06:
Is this issue still open? I am looking for a good first issue to pick up for contributing to uTensor project.

dboyliao, 2020-12-07:
@balajikulkarni 
This issue has been closed by #216.
I should close this issue.
Thanks for pointing it out.

closed 2020-12-07 by dboyliao