uTensor/uTensor #219: Models interface should push and pop default allocator context on eval

mbartling, 2020-11-30:
To support multiple allocator contexts in the tensor destructor, model interface should push and pop current allocator pointers to a queue in the default current context else risks failed destruction

mbartling, 2020-11-30:
@dboyliao found it

dboyliao, 2020-11-30:
Example code to reproduce the error:
https://github.com/dboyliao/PyDataTP_2020/blob/master/4_model_interface/model_interface.cpp

dboyliao, 2020-11-30:
As a side note, the main issue is that we need to switch the ram & meta allocator back to those in the user space after model eval.
Since model eval will temporary switch allocators to the ones inside the model, not the ones in user space.

If we forget to switch allocators back, tensor deconstructor will fail.