uTensor/uTensor #225: not building because of arenaAllocator.cpp error

acilione, 2021-11-05:
// Return the amount of free space at the tail
uint32_t localCircularArenaAllocatorBase::tail_capacity(){};

mbartling, 2021-11-08:
Hi @acilione, this method is mostly meant for some advanced integration tests (that I haven't gotten around to yet). You should be able to get by with having the method return 0 for now.

-Michael

acilione, 2021-11-09:
Thank you, I got it solved. Just for info, it gave me another error:
[Error] ReduceFunc.hpp@22,57: no matching constructor for initialization of 'uTensor::ReferenceOperators::ReduceOperator'
[Error] ReduceFunc.hpp@32,19: '_dims' is a private member of 'uTensor::ReferenceOperators::ReduceOperator'
[ERROR] In file included from ./uTensor/src/uTensor/ops/ReduceFunc.cpp:1:

but I managed to run correctly the hello world example just by commenting out the ReduceFunc.hpp and ReduceFunc.cpp files.

mbartling, 2021-11-09:
Weird, I haven't seen this crop up in the past. Which compiler and target are you using? Am going to try to reproduce it.

acilione, 2021-11-09:
I'm using DISCO_L475VG_IOT01A and ARMC6.
Hope this helps

acilione, 2021-11-09:
> Weird, I haven't seen this crop up in the past. Which compiler and target are you using? Am going to try to reproduce it.

I now managed to get it to compile correctly without commenting out all the file, with the following fixes:
in ReduceFunc.hpp:

/***original code***/
class ReduceOperator : public OperatorInterface<2, 1> {
 public:
  ReduceOperator(initializer_list<uint16_t> dims);
  uint32_t adjust_linear_idx(Tensor& tensor, uint32_t idx);

 private:
  uint16_t _dims[4];
};
/****************/

/***fixed code***/
class ReduceOperator : public OperatorInterface<2, 1> {
 public:
  ReduceOperator(initializer_list<uint32_t> dims);
  uint32_t adjust_linear_idx(Tensor& tensor, uint32_t idx);

  uint16_t _dims[4];
};
/*****************/
changed  
ReduceOperator(initializer_list<uint16_t> dims);    to  
ReduceOperator(initializer_list<uint32_t> dims); 

made  uint16_t _dims[4];   public

  ----------------------------------------
 
 in ReduceFunc.cpp:
  
  /****original code****/
  namespace uTensor {
namespace ReferenceOperators {
ReduceOperator::ReduceOperator(initializer_list<uint16_t> dims) {
  size_t i = 0;
  for (auto s : dims) {
    _dims[i++] = s;
  }
}
/**************************/

/****fixed code****/
  namespace uTensor {
namespace ReferenceOperators {
ReduceOperator::ReduceOperator(initializer_list<uint32_t> dims) {
  size_t i = 0;
  for (auto s : dims) {
    _dims[i++] = s;
  }
}
/*************************/
changed 
ReduceOperator::ReduceOperator(initializer_list<uint16_t> dims)   to
ReduceOperator::ReduceOperator(initializer_list<uint32_t> dims)

closed 2021-11-20 by acilione