uTensor/uTensor #245: Compatibility with only C based project (no c++ support)

asmitaj08, 2025-03-02:
Hello,
I wanted to know if it is possible to use uTensor with one of the nrfsdk projects (older one - [Link](https://www.nordicsemi.com/Products/Development-software/nRF5-SDK/Download#infotabs)) which only supports C (no c++). All the examples related to uTensor which I found, is generationg c++ files. I would really appreciate some suggestions.
Thank you!

dboyliao, 2025-03-03:
My suggestion will be compile uTensor as static library and writing C wrappers around C++ api to integrate with a C only project.
As far as I know, we don't have an example on C-only project for now.
Let us know if you need more help.

asmitaj08, 2025-03-03:
Got it, Thank you!